Martinsbl/nrf5-mpu-examples #3: Support for ICM-20601

kbystedt, 2016-11-07:
How difficult is to get support for the ICM-20601. I am new to embedded project,

Martinsbl, 2016-11-08:
It is all about the register map and the configuration bits. I had a quick look at the ICM's register map and it looked very similar, but unfortunately not identical, to the MPU9255 register map. So what you would have to do is to make a register map header file like the three header files included in this repository. Then you need to go through the configuration registers and possibly add ICM specific parameters to the configuration structs in app_mpu.h. Unfortunately I don't have the time to do this in any foreseeable future, but feel free to try for yourself.

khalil50, 2017-06-15:
HI Martinsbl , i m working with mpu 6050 and a nrf52832 pca 10040 card, i use  your code. but i didnt succeed to get it work wel. i configure the target by putting define MPU60x0 My sda and scl is pin 26 and pin 27 but in your library were made 3 and 4 ! so can u help me to configure your code please