Martinsbl/nrf5-mpu-examples #4: Problem with Magnetometer-Access

RayBreslin, 2016-11-09:
I have tried out all your examples and with some modifications they work quite well for my custom-board with a nRF52832- and a MPU9250-chip even within an Eclipse-GCC-toolchain. Currently I can access and print out all data coming for accelerometer, gyroscope and temperature. 

But now comes the hard part of also accessing the magnetometer-data via I2C-communication. Apparently I can verify that both sensors of the MPU9250 are present on the I2C-bus with their corresponding I2C-address (0x68 and 0x0C) using the twi-scanner-example from the Nordic-SDK. I also ensured that I am not having any hardware-problems since the same MPU9250 is working just fine on different development-boards (f.ex. Teensy 3.2/CSR1010). 

Here is what I did so far:

First of all I copied and modified the mpu_read_register()- and mpu_write_register()-functions from nrf_drv_mpu_twi.c with the AK8963-I2C-address. Then I tried to configure the operation-mode (8 Hz or 100 Hz) of the AK8963 via the REG_CNTL_1-register - just as I did it on the other development-boards.
When I try to readout data from the REG_ST1-register (not REG_MAGN_XOUT_L) then I am getting no output at all.

My assumption is that I either need some additional twi-initialization for the communication of a 2nd-I2C-address or I might need some strong pull-up-option in order to provide more power for the twi-interface (was necessary with my other development-boards).

Do you happen to have any ideas on what is necessary to actually have 2 or more devices communicating via I2C with the nRF52832? 

Any help is very much appreciated...

Martinsbl, 2016-11-11:
Hi. I created a new branch called magnetometer. I it is a very quick and dirty test only tested with PCA10040, twi and mpu9255. It enables the magnetometer and sets it in continuous measurement mode 2. It seems like you have to be careful about which registers you read out. If I read register 0-8 it seems to work. If I read all registers in one go it does not work. Check out nrf5-mpu-magnetometer. It works for me.

RayBreslin, 2016-11-11:
Oh thank you very much for your efforts - in the meantime I also managed to set up a working example for the nRF52832 in combination with the MPU9250! I have not checked all of its functionality though, but I dont expect any further difficulties...

Just out of interest I tried to look for your "nrf5-mpu-magnetometer"-path but I could not find it... (I hope I am not blind)

Martinsbl, 2016-11-12:
Did you check out the magnetometer branch? https://github.com/Martinsbl/nrf5-mpu-examples/tree/magnetometer

RayBreslin, 2016-11-14:
Oh I must have missed it somehow! When I checked all available branches the last time I couldnt see it...

But its all fine now! :)

closed 2016-11-14 by RayBreslin