Martinsbl/nrf5-mpu-examples #5: Missing include-files for nrf-ble-mpu-simple-example

RayBreslin, 2016-11-14:
Hello,

I was trying to set up the example "nrf-ble-mpu-simple" by including all relevant header- and source-files from the nordic-SDK (v12), but unfortunately I couldnt find any app_trace.h-/app_trace.c- and device_manager.h-/device_manager.c-files.

Were they only available in previous SDKs?

I would very much appreciate any help you can provide!

Regards Ray

Martinsbl, 2016-11-14:
In SDK 12 the app_trace library was replaced with the Logger Module http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v12.0.0/nrf_log.html?cp=4_0_0_3_15
It is quite advanced compared to app_trace so porting might take some time. Unfortunately I don't have the time to look into it in the foreseeable future.

RayBreslin, 2016-11-14:
Thank you for that information - I kind of assumed that!

What about the device_manager-files? Have they been replaced too?

Martinsbl, 2016-11-15:
Yes. The Peer Manager replaces the previous Device Manager. http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v12.0.0/lib_peer_manager.html?cp=4_0_0_3_1_8

RayBreslin, 2016-11-16:
Ah ok got it - example is running properly now with just minor modifications!

Thanks again! :)

closed 2016-11-16 by RayBreslin

kbystedt, 2016-11-16:
Will you upload modifications?

RayBreslin, 2016-11-16:
I could ofc, but I dont know if that actually helps, since its just an adaption for SDK v12.0+!

First I replaced the app_trace- and device_manager-functionality by the ones which Martin mentioned, but in my endversion I basically just removed all links and usages for either nrf_log and peer_manager since I had no use for them currently.