Martinsbl/nrf5-mpu-examples #8: Porting from Keil to Segger Embedded Studio

haarislabs, 2018-07-18:
I followed the steps provided by Nordic : http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.gs.ses%2Fdita%2Fgs%2Fdeveloping%2Fimport_ses.html
To port your examples from keil to Segger. I first tried to port with nrf-mpu-simple and it worked fine. When I ported nrf-ble-mpu-simple it give an error that `undefined reference to __app_ram_start__ `  .
I know that the error is coming due to the "flash_placement.xml" file. I'm a newbie so I don't know much about what goes on in that. Can you help me out in solving this problem, any kind of guidance would be helpful.

closed 2018-07-19 by haarislabs

gjfwalker, 2018-09-25:
Did you solve this issue? If so, how?

haarism19, 2018-09-27:
I didn't port it directly as shown in the tutorial instead, I picked a base example with segger studio as my IDE (which in my case is ble_app_hrs_freertos) and added these libraries on top of that. Had to do a few changes in "sdk_config.h" and add the files into my working directory that's all. It works fine after that.

haarislabs, 2018-09-27:
I didn't port it directly as shown in the tutorial instead, I picked a base example with segger studio as my IDE (which in my case is ble_app_hrs_freertos) and added these libraries on top of that. Had to do a few changes in "sdk_config.h" and add the files into my working directory that's all. It works fine after that.