Martinsbl/nrf5-mpu-examples #9: bug with nrf5-ble-uart-data-ready-interrupts?

austinfennacy, 2018-09-16:
As a preface, since BLE examples no longer compile to a size smaller than the code size limitation of Keil at 32kB after SDK 13, I'm working with this project's 3rd release (April 2017, MPUxxxx examples for SDK11). I know I could theoretically use later examples but I've had a miserable time trying to adapt Kiel projects to Segger or Eclipse+Makefile to compile larger file sizes.

I'm trying to use `nrf5-ble-uart-data-ready-interrupts` to send MPU60x0 data over a BLE UART connection.  Using an MPU6050, nRF52 DK, and iPhone 5s. I've successfully gotten nrf5-mpu-data-ready-interrupts to work with a wired UART connection so I know the nRF<->MPU connection is okay (side note, I did catch how for this release, MPU_MPU_INT_PIN is defined as 5 for the wired UART but defined as 30 for BLE UART). When I use nrf5-mpu-data-ready-interrupts, I witness very regular looking signals on the MPU_SCL, MPU_SDA, and MPU_INT lines when observing via an oscilloscope. I have independently gotten the `ble_app_uart_s132_pca10040.hex` example to work with my setup.

However, when I build and upload ` nrf5-ble-uart-data-ready-interrupts` using uVision and nRFgo using the MPU60x0 define in C/C++ Target Options, things goes wrong. When looking in the Serial monitor, the monitor prints out the first few characters of the printf statement (i.e. "MPU BLE UA") and the number of characters printed but it never gets past the whole line. This occurs before BLE even gets involved in the code (see below).. And when watching the MPU_SCL line on an oscilloscope, there's some clock signal right after the nRF52 DK is reset, which gives me hope because at least some communication attempt is made, but at the same time should the `mpu_setup()` line begin executing before the `printf(...)` line stops printing? I've tried commenting out all `printf(...)` lines because I know later versions use `NRF_LOG_INFO()` so perhaps `printf(...)` is the problem but that does not resolve the issue.

```
int main(void)
{
    uint32_t err_code;
    bool erase_bonds;
    LEDS_CONFIGURE(LEDS_MASK);
	  LEDS_OFF(LEDS_MASK);

    // Initialize.
    APP_TIMER_INIT(APP_TIMER_PRESCALER, APP_TIMER_OP_QUEUE_SIZE, false);
    uart_init();
    printf("MPU BLE UART example. Compiled @ %s.\r\n", __TIME__);
    mpu_setup();
    ...

```

Has anyone else encountered this issue? Any tips on how to get this code working, or otherwise get the MPU to send data via BLE?

prasanpro, 2019-01-16:
I'm using keil project version in SDK14 to get some data over UART. I didn't exactly did the same things but tried to print MPU6050 data with printf, but surpringly it stopped all code flow. Currently using keil, but wanted to use Segger to send the data over BLE. Are you using Keil or Segger?

austinfennacy, 2019-01-17:
I use Segger these days since you get a free full licence with no restrictions.
One thing I noticed for Segger is that printf statements are turned off by default, not sure why but probably for optimization reasons? Anyhow in Segger, go to "project options", search for "print", and enable everything. There might be something similar for Keil.

That said, I never got this repo to work for me. I moved on to doing from "scratch" using Segger, SDK 15.0.0,  TWI drivers, and [this tutorial](https://devzone.nordicsemi.com/tutorials/b/bluetooth-low-energy/posts/ble-characteristics-a-beginners-tutorial).

If you'd like to get this working, my advice is to

1. get TWI Scanner to work and understand the code snippets that write to the MPU6050's "WHO_AM_I" register using the TWI driver
2. modify this file so instead of looping through all addresses, you only select the MPU6050's address
3. skim the [advertising tutorial](https://devzone.nordicsemi.com/tutorials/b/bluetooth-low-energy/posts/ble-advertising-a-beginners-tutorial) and the [services tutorial](https://devzone.nordicsemi.com/tutorials/b/bluetooth-low-energy/posts/ble-services-a-beginners-tutorial)
4. follow every step in the [characteristics tutorial](https://devzone.nordicsemi.com/tutorials/b/bluetooth-low-energy/posts/ble-characteristics-a-beginners-tutorial) and get it to work
5. change the timeout event handler function in the characteristics tutorial to run the TWI code you used in step 2 (CAVEAT - you'll need to make sure you copy+paste all dependent #imports, #defines, and functions from the TWI code. after that, you'll need to change your "sdk_config.h" file in the local project directory, look up how to get CMSIS working for segger to edit these options, and then make sure you enable the relevant flags in the TWI diver folder. Just check everything that says TWI in the scanner example and make sure those flags are checked in the characteristic code, otherwise you'll get compile errors about something missing)
6. look at some arduino library and how they first initialize the MPU6050 by writing config settings to registers and recreate that in C, most code should be copy+pasteable and you'll have to use the nRF TWI drivers in place of ardruino wire library functions. after that, do the same for reading data from MPU6050 registers. I would base work off of code you can get to compile, which is why I reference the arduino version, but perhaps you could get some insight from this code as well.

Lot of steps but that's what I did to get this working for a different accel/gyro IC. gl!